FreeCAD/freecad.workbench_starterkit #2: Discussion in FreeCAD-addons about using metadata.txt the way QGIS does

luzpaz, 2017-11-10:
https://github.com/FreeCAD/FreeCAD-addons/issues/29#issuecomment-303949382
Is this something akin to the idea of `Manifest.in` ?

looooo, 2019-12-19:
I guess this comment is somehow related.
https://forum.freecadweb.org/viewtopic.php?f=8&t=40749&start=20#p354989

Btw.: The Manifest.in is a list of additional files which should be packaged.

midgetfc, 2019-12-19:
Hello,
first of all, sorry to intrude in this discussion.
I made the comment quoted above and I'm not sure it is related.

I maybe mistaken, but @luzpaz is asking for a way to describe dependencies of an addon, and I think the mechanism is already provided by setuptools for this kind of 'new style pip installable addon', if these dependencies are available for download.
(see https://setuptools.readthedocs.io/en/latest/setuptools.html#declaring-dependencies)

As @looooo said, the classical use for Manifest.in is to include in the distribution additional files not readily identified as 'code' files, like resources or supplementary documentation.

luzpaz, 2021-12-18:
This is being superceded by @chennes's work on the Addon Manager Package Metadata: https://wiki.freecad.org/Package_Metadata

chennes, 2025-09-28:
And indeed, though the Addon Manager will happily read your metadata.txt file and mine it for dependencies, the preferred, and officially-supported, dependency mechanism is via the package.xml metadata file. This starterkit does not yet produce that file, however.

closed 2025-09-28 by chennes