FreeCAD/freecad.workbench_starterkit #9: Workbench not loading

CyrilWaechter, 2019-12-27:
Following commit 59fb144, the workbench is not loading anymore. It looks weird to me : 
1. If you have pyrate or FCGear you will not notice the issue as it contains a [`__init__.py`](https://github.com/mess42/pyrate/blob/master/freecad/__init__.py) with `__path__ = __import__('pkgutil').extend_path(__path__, __name__)` inside
2. If you have no workbench built this way using this `__init__.py` file. The workbench will not load.

looooo, 2019-12-27:
This file should not exist in any workbench because if it does exist multiple workbenches /modules can possibly install and overwrite the same file (freecad/__init__.py). I deleted this file already for the gear-workbench (development-branch).
The problem is that if this file exist in one workbench it's possible another workbench is not added to the path.

Btw.: can you share your FreeCAD info?

CyrilWaechter, 2019-12-28:
> Btw.: can you share your FreeCAD info?

Sure, I am currently using the AppImage as there is an issue between python 3.8 and shiboken : 
```
OS: Manjaro Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19020 (Git) AppImage
Build type: Release
Branch: master
Hash: 04b54cd6a5f1962f4e40ae09e86507317da35a88
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/Switzerland (fr_CH)
```
> The problem is that if this file exist in one workbench it's possible another workbench is not added to the path.

If the __init__.py file is in each workbench, no issue, it works (on my machine at least) it fails only if the __init__.py file is nowhere.

looooo, 2019-12-28:
@CyrilWaechter Can you try the latest version (0.19.19093). This should include some changes which should fix this issue (hopefully). I am not totally sure but with these changes, it should not be a problem if workbenches include the `freecad/__init__.py` file or not if installed to the .FreeCAD/Mod directory. It would be nice if you can prove this assumption.

looooo, 2019-12-28:
The problem you are facing is related to some tests I did on getting freecad importable from python directly (`import freecad`). The file used to initialize freecad in (0.19.19020) can be found here:
https://github.com/looooo/freecad.python/blob/0.0.3/freecad/__init__.py.template
As you can see the path extension is missing here and therefor modules not containing this file will not be importable. For (0.19.19093) I am now going for a different approach (patching the freecad-source: https://github.com/looooo/freecad-feedstock/blob/4bc77699f9c2cbfa9d76cd1b11e6310a0520106a/recipe/freecad-sitepackges.patch) If this works out, I will try to create a PR for FreeCAD and write a proposal how to further improve the situation with the new import structure and new-style workbenches.

I know it is very frustrating depending on something which someone else is experimenting with. But these experiments are necessary to finalize the concept of the new import structure and make freecad importable from python directly. So I am very thankful for this feedback.

CyrilWaechter, 2019-12-28:
> Can you try the latest version (0.19.19093).

I confirm it worked for me : 
```
OS: Manjaro Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19093 (Git) AppImage
Build type: Release
Branch: master
Hash: 9e4710d679d44613a1ae4264b86ef9f6d12f4e5a
Python version: 3.7.6
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/Switzerland (fr_CH)
```
> I know it is very frustrating depending on something which someone else is experimenting with. But these experiments are necessary to finalize the concept of the new import structure and make freecad importable from python directly. So I am very thankful for this feedback.

Current situation is confusing because wiki page describe another way. As soon as you get a stable solution tell me. I will be happy to help documenting the process. I think this is very important for potential new developers to not face this kind of confusion during their first steps.

closed 2019-12-28 by CyrilWaechter