FreeCAD/freecad.workbench_starterkit #10: Document Motivation Behind "New" Workbench Style

gbroques, 2020-01-25:
There's a nice mention about the "old" style of workbenches in the **README** under **Additional Information**:

> old_style_module: A package which is plugged into FreeCAD by adding it's base-directory to sys.path and uses **Init.py** and **InitGui.py** to get initialized by FreeCAD.

However, it's not crystal-clear to me **why** you would want to switch from the "old" style to the "new" style.

What's the motivation to switch, and key differences between both styles?

@looooo Has discussed this disparately throughout the forum, but it'd be nice to have a concise and clear article somewhere.

---

I'm currently involved in a project where I have to support FreeCAD 16 workbenches, and there's resistance behind upgrading to a later FreeCAD version.

Having more information about the advantages of the "new" style would be very helpful.

looooo, 2020-01-28:
We cannot fully move towards new-style because of python2-support is still ongoing. The problem here is a combination of addon-manager / namespace-packages / how freecad is maintained for several distros. There is a little difference between py2 namespace-packages and py3-namespace packages.

You already mentioned the main cons and pros. There are also other pros (like using `__file__` in a iinit_gui.py which is caused by using `exec` to initialize the workbenches and using `exec` for this task is known to be bad-practise ...)

looooo, 2020-03-10:
@gbroques can you make a PR and add the PROS. The incompatibility with older freecad versions was already added.

gbroques, 2020-03-17:
@looooo Sure thing, not a problem.

I should be able to toss something up within the next day or two.

commit e7649ddd (2020-03-18): Document motivation behind 'new style' modules (Closes #10)
closed 2020-03-18 by looooo