FreeCAD/freecad.workbench_starterkit #13: Better Terminology Than Old vs. New Workbenches

gbroques, 2020-06-10:
From the **README**:

> There are currently two FreeCAD workbench styles. For lack of better names, we use the terms "old style" and "new style" (See Glossary terms used in this discussion). For more discussion about the motivation behind the "new style" see Motivation for New Style Modules.

Let's come up with better names than "old" vs. "new". Those names only describe the temporal aspects of each workbench style (i.e. one is traditional usage, and one is a pattern this template advocates for adoptions).

Maybe "Modern Workbenches" or "Packageable Workbenches" is better than "new workbenches"?

One of these proposed terms seem to encapsulate the benefits of the "new style":
https://github.com/FreeCAD/freecad.workbench_starterkit#motivation-for-new-style-modules

All of the benefits are things you'd expect from *modern* or *packageable* software.

Are there any word-smiths out there with better suggestions for names?

gbroques, 2020-06-10:
Vocx suggsted "namespace" workbenches on the forum:
https://forum.freecadweb.org/viewtopic.php?f=22&t=47460#p407171

gbroques, 2020-06-14:
mnesarco proposed "Legacy Workbenches" as opposed to "old":
https://forum.freecadweb.org/viewtopic.php?f=22&t=47460#p407530

loo thinks "legacy" / "namespace" workbenches might be a good name pair:
https://forum.freecadweb.org/viewtopic.php?f=22&t=47460#p407555

gbroques, 2020-06-15:
Looks like the thread has focused on the terms "Legacy" and "Standard" workbenches:
https://forum.freecadweb.org/viewtopic.php?f=22&t=47460&start=10#p408590

This implies having a set of FreeCAD workbench standards -- which is now what the thread is focused on.

looooo, 2020-07-14:
Not sure if we came to any agreement, but I think legacy / namespace-workbench the best option. Is it ok, to modify the docs to match this proposal?

gbroques, 2020-07-15:
> Not sure if we came to any agreement, but I think legacy / namespace-workbench the best option. Is it ok, to modify the docs to match this proposal?

Yes, I think that's an improvement upon the terminology of "old" vs. "new".

looooo, 2020-07-16:
https://github.com/FreeCAD/freecad.workbench_starterkit/commit/821cd287ef30fd6a0ba7cb420b6b7e6ff1b91577

gbroques, 2020-07-16:
> [821cd28](https://github.com/FreeCAD/freecad.workbench_starterkit/commit/821cd287ef30fd6a0ba7cb420b6b7e6ff1b91577)

This looks great @looooo! Thank you for your efforts.

I'm closing this.

closed 2020-07-16 by gbroques